jackal/jackal_robot #11: Whatsup with ros kinetic support?

TankPi, 2017-01-11:
At my company we recently migrated to kinetic. Would be nice if we could do that for our jackel too.

mcoteCPR, 2017-01-17:
Hi @TankPi 

Thanks for taking interest in using Jackal in Kinectic! We're certainly planing on updating all our robots to ROS kinetic, however unfortunately at this point we're not able to provide a timeline. Keep a close eye on this Github repository, and I'll let you know when we have something that you can test out on your own unit.

dniewinski, 2018-03-19:
Sorry for the delay on this. Kinetic is currently in development and we are are hoping to release it within the next month or two. This issue will be updated when that is completed.

dniewinski, 2018-06-09:
Despite many delays, we have finally finished and released support for Kinetic! Many of the packages should be available through http://packages.ros.org or from http://packages.clearpathrobotics.com

closed 2018-06-09 by dniewinski

dniewinski, 2018-06-11:
It's definitely still a thing.  Thanks for pointing this out. I'll have a look into why RB isn't there