jackal/jackal_robot #13: Change in IMU message type

hawesie, 2018-09-11:
Running under Kinetic I see the following message from `roslaunch jackal_base base.launch `:

```
[ERROR] [1536651743.105581726]: Client [/imu_filter] wants topic /imu/mag to have datatype/md5sum [sensor_msgs/MagneticField/2f3b0b43eed0c9501de0fa3ff89a45aa], but our version has [geometry_msgs/Vector3Stamped/7b324c7325e683bf02a9b14b01090ec7]. Dropping connection.
```

This is due to a change in [imu_filter_madgwick](http://wiki.ros.org/imu_filter_madgwick#Supported_devices):

```
~use_magnetic_field_msg (bool, default: false for Hydro and Indigo, true for Jade+)

    If set to true, subscribe to the /imu/mag topic as a sensor_msgs/MagneticField; if set to false (deprecated), use geometry_msgs/Vector3Stamped. 
```

Therefore for ROS distros below Jade `base.launch` requires the following added to the `imu_filter` node configuration:
```
          <param name="use_magnetic_field_msg" value="false" />
```

hawesie, 2018-09-11:
Whilst I've created a PR I'm a bit unsure if this is actually the correct thing to do, since Kinetic > Jade, which seems to conflict with the docs.

TYXDG, 2022-04-28:
Hi, I've also been using jackal for GPS/IMU/ Odometry fusion location. Navsat_transform_node requires the absolute orientation provided by IMU, but imu_filter_Madgwick's use_magnetic_field_msg has been discarded. So is there any way to change the type of the/imu/mag messages published by Jackal to sensor_msgs /MagneticField? Or is there another package for sgeometry_msgs/Vector3Stamped message types? In addition, the three components of my jackal/imu/mag are all 0 indoors, and they are all 10-5 or 10-6 in the outdoor. I guess there seems to be some problems with the magnetometer reading. Could you please give me some help? thank you

tonybaltovski, 2022-04-28:
@TYXDG which ROS distro are you using?  Also, which version of jackal_firmware do you have installed?  Can be checked by echoing the `/status` topic.

TYXDG, 2022-04-28:
@tonybaltovski thanks for your reply! It's embarrassing to say, but I'm a new user of Jackal, so I don't know how to check the version. Attached is the /status topic recorded after roslaunch jackal_base base. launch

header: 
  seq: 0
  stamp: 
    secs: 1651126124
    nsecs: 949279789
  frame_id: "base_link"
hardware_id: "0.3.9"
mcu_uptime: 
  secs: 152
  nsecs:         0
connection_uptime: 
  secs: 0
  nsecs:         0
drivers_active: True
driver_external_stop_present: False
driver_external_stop_stopped: False
measured_battery: 27.2449245453
measured_12v: 11.9396343231
measured_5v: 5.01569652557
drive_current: 0.683917880058
user_current: 1.98189902306
computer_current: 0.00311243324541
total_current: 2.66892933846
total_current_peak: -1.0
total_power_consumed: -1.0
---
header: 
  seq: 0
  stamp: 
    secs: 1651126125
    nsecs: 949279789
  frame_id: "base_link"
hardware_id: "0.3.9"
mcu_uptime: 
  secs: 153
  nsecs:         0
connection_uptime: 
  secs: 0
  nsecs:         0
drivers_active: True
driver_external_stop_present: False
driver_external_stop_stopped: False
measured_battery: 27.2622070312
measured_12v: 11.9386615753
measured_5v: 5.01621580124
drive_current: 0.749266088009
user_current: 1.85299503803
computer_current: 0.00293458020315
total_current: 2.60519576073
total_current_peak: -1.0
total_power_consumed: -1.0
---

tonybaltovski, 2022-04-28:
@TYXDG How about the ROS distro?  Run `echo $ROS_DISTRO` to check.  If you are using melodic or newer, you want to upgrade your Jackal's firmware to at least `0.4.0`.  See here for instructions: https://www.clearpathrobotics.com/assets/guides/melodic/jackal/update.html#:~:text=In%20the%20top-middle%20of,ready%20to%20receive%20new%20firmware.

TYXDG, 2022-05-05:
I'm sorry for replying to you after such a long time. Currently the ROS version installed on jackal is melodic, does that mean I can skip "GETTING NEW PACKAGES" and start directly with "MCU FIRMWARE UPDATE"? In addition, there seems to be no Jackal_firmware package on my Jackal. May I ask how to operate in this case? thank you

tonybaltovski, 2022-05-05:
@TYXDG Update your packages first.  Are you sure then isn't?  What is the output of `rosversion jackal_firmware`?

TYXDG, 2022-05-05:
the output of rosversion jackal_firmware is Cannot locate [jackal_firmware]

tonybaltovski, 2022-05-05:
Is this on the robot or your own laptop?  Regardless, you can install it using `sudo apt install ros-$ROS_DISTRO-jackal-firmware` and then follow: https://www.clearpathrobotics.com/assets/guides/melodic/jackal/update.html#:~:text=In%20the%20top-middle%20of,ready%20to%20receive%20new%20firmware.

TYXDG, 2022-05-06:
I use SSH command to remotely log in the IP of jackal's internal computer on my computer. Then I type 
"sudo apt install ros-melodic-jackal-firmware"
under jackal's IP and the output is "E: Unable to locate package ros-melodic-jackal-firmware". Then I googled the Jackal_firmware package and it didn't seem to exist (http://wiki.ros.org/jackal_firmware), just a Windows version (https://github.com/jackal/jackal_firmware_ms).

TYXDG, 2022-05-16:
Hello, do you have any solution to the above problem?  Thank you very much

civerachb-cpr, 2022-05-16:
The Jackal firmware package is closed sourcr and not available on GitHub. To install it you'll need add Clearpath's apt servers, following the instructions here: https://packages.clearpathrobotics.com

This is done automatically when installing the OS on the Jackal when using Clearpath's ISOs from https://packages.clearpathrobotics.com/stable/images/latest/, but must be done manually if you're using a standard Ubuntu image from Canonical.

TYXDG, 2022-05-16:
According to the https://packages.clearpathrobotics.com, I have manually completed operations, but when I type "rosversion jackal_firmware", The output is still "Cannot locate [jackal_firmware]". Is there any way to check if I have firmware installed

TYXDG, 2022-05-17:
Thanks very much for your help, now the version of /status and rosversion Jackal_firmware are all 0.4.3, and the message type has changed to sensor_msgs /MagneticField. But now there is another problem, the magnetic_field has a reading of [X, Y, Z]=[2.65e-5, -1.81e-6, 5.81e-6], and this reading does not change when I rotate the car in place with the handle. Can you offer some solutions to this problem? Thank you again

civerachb-cpr, 2022-05-17:
The IMU in the Jackal is located in the back, very close to the PC, power cables, and one of the motors.  This results in some pretty strong magnetic fields, and makes the magnetometer data unreliable.  We normally leave the magnetometer disabled for this reason.

You're welcome to try calibrating it using the guide [here](http://www.clearpathrobotics.com/assets/guides/melodic/jackal/calibration.html), but the performance will likely never be very good because of the location of the sensor.