jackal/jackal_robot #15: Missing Melodic support

rgreid, 2019-11-19:
Hi, do we have a timeline for Melodic support for Jackal robots?

tonybaltovski, 2019-11-20:
Hi @rgreid, we are targeting the end of the month or sooner.  I will update this ticket when it is ready.

rgreid, 2019-11-22:
@tonybaltovski I emailed you and submitted a Clearpath Support request just now. Please let us know if/how we can test a pre-release of your Melodic support. Thanks

tonybaltovski, 2019-12-11:
@rgreid it has been released.  You can grab the [ISO](https://packages.clearpathrobotics.com/stable/images/0.4.1/) and a firmware update is needed.

closed 2019-12-11 by tonybaltovski

rgreid, 2019-12-11:
Thanks heaps. I found new debs on your [repo](https://packages.clearpathrobotics.com/stable/ubuntu/pool/main/r/ros-melodic-jackal-base/), however this repo is still on Kinetic. Wondering where the updated sources are? 

We're running a custom firmware on our Jackals that adds a remote E-stop. Are the firmware changes from v0.3.9 to v0.4.0 required or cosmetic?

tonybaltovski, 2019-12-11:
There are no changes between the versions so there is only a kinetic-devel branch.

I am not sure about the custom firmware.  The only change was switching from geometry_msgs/Vector3Stamped to sensor_msgs/MagneticField.

tonybaltovski, 2019-12-11:
Does your jackal report `0.3.9` for `jackal_msgs/Status/hardware_id`?

rgreid, 2019-12-11:
Yes, we have many Jackals and are testing on both `0.3.9` unmodified, and a custom firmware.  

I have a CP [support request](https://support.clearpathrobotics.com/hc/en-us/requests/6776) outstanding that I opened a few weeks ago, where I outlined some issues we've been having with the Jackal base running the `kinetic-devel` branch on Melodic already. Could you please look this one up, and/or I'll open another issue here. Thanks.

YiXuan157, 2021-09-15:
Hello,
In official webpage 
https://clearpathrobotics.com/assets/guides/kinetic/kinetic-to-melodic/upgrade-os.html
The link of downloading the img is empty for jackal-melodic
http://packages.clearpathrobotics.com/stable/images/latest/

May I ask https://packages.clearpathrobotics.com/stable/images/0.4.1/melodic-jackal/amd64/ you provide in this issue is the last stable version?

civerachb-cpr, 2021-09-15:
The latest ISO we use is here: https://packages.clearpathrobotics.com/stable/images/latest/melodic-bionic/amd64/
The previous single-robot ISOs have been discontinued; the new one adds a prompt asking what robot you're installing, allowing us to support all of our platforms with a single ISO.
Unfortunately due to a server-side glitch our update script is still creating the folders for the older ISOs, even though they aren't being built anymore.  I've got an open ticket with our IT department to get that resolved, so hopefully we can alleviate the confusion soon.