jackal/jackal_robot #16: Ability to change "imu_link" frame id?

aamirhatim, 2020-01-03:
Apologies if I'm in the wrong area, but I'm looking to add a tf_prefix to the published imu data and I can't seem to find where the frame_id is even being set. When `jackal_node` is running, I can see that it is publishing imu data to the `madgwick_filter` with a frame id name of "imu_link". The `madgwick_filter` publishes a filtered imu message with the same frame_id name. Looking at the `madgwick_filter` source code, it simply takes the frame_id from the incoming raw topic and pastes it in the output filtered topic. So I can only assume the frame id is being set somewhere in `jackal_node`. I've looked through this repo but cannot find any mention of this frame_id being set, so any suggestion on where else I can look would be a huge help.
Looking at it bigger picture - I'm simply trying to add a namespace to the frames so that I can start multiple instances of this node and not have a generic "imu_link" name be used.

Things I've tried: 
1. Starting the node in my desired namespace
2. Set parameters like tf_prefix, frame_id, imu_link, etc. within `jackal_node` namespace
3. Manually set the frame id in madgwick - this "worked" but it involved changing the source code to read in a param I would set in the parameter server.

My system:
I'm running ROS Kinetic/Ubuntu 16.04

Thank you in advance!

mikepurvis, 2020-01-04:
The frame_id for the raw IMU data originates in firmware, which is where that topic comes from, via rosserial. So a firmware update would be necessary to change it there unless you don't mind hacking the madgwick filter.

The reason this has never come up before is that 99% of tf_prefix usage is when simulating multiple robots under the same Gazebo instance and ROS Master. In that scenario, it doesn't matter because the madgwick filter isn't in the picture at all— the imu data is generated by a tf_prefix-aware Gazebo plugin.

Are you trying to run multiple physical Jackals on a single ROS master?

aamirhatim, 2020-01-06:
I see, that makes a lot more sense then. I'm planning on running multiple physical robots, but not necessarily all of them Jackals. Everything in Gazebo works fine with the proper namespaces being used, as you mentioned. We currently have one Jackal and one Ridgeback and will be running them under one ROS Master.

The Madgwick fix wasn't anything too in the weeds, so that may be sufficient. If you have any other recommendations, I'd be open to hearing them. Otherwise feel free to close the issue - I really appreciate the help!

mikepurvis, 2020-01-06:
Sounds great! Good luck with your project.

closed 2020-01-06 by mikepurvis