jackal/jackal_robot #28: Network Issues when setting up VLP-16 on Jackal

NithishkumarS, 2021-03-12:
I have been trying to setup velodyne VLP-16 on jackal. Since by default both the static and dhcp port point to the same IP address, I followed this tutorial to configure my networks.
https://answers.ros.org/question/264413/how-can-i-get-a-clearpath-jackal-to-receive-hdl-32-data/

I assumed my network is separated. So, I use the first port (static) to connect a laptop that runs perception and navigation algorithm. I have setup the ROS master IP as per the instructions in the tutorials. So, by default once I connect I would be able to see the ros topics published without sshing into the Jackal. 

I used the second ethernet port to connect the lidar and once I add the route, I am unable to connect to the ros master in the Ubuntu PC connected to the first port. 
That prevents me from running the filters on the Point cloud data.

I have attached my /etc/network/interfaces file. Kindly provide me a solution

```
auto lo br0 br0:0
iface lo inet loopback

# Bridge together physical ports on machine, assign standard Clearpath Robot IP.
iface br0 inet static
  bridge_ports regex eno1
  address 192.168.131.1
  netmask 255.255.255.0
  bridge_maxwait 0

#auto wlan0
#iface wlan0 inet dhcp
#                wpa-ssid NETGEAR21-5G
#                wpa-psk phobicriver918

# Also seek out DHCP IP on those ports, for the sake of easily getting online,
# maintenance, ethernet radio support, etc.
iface br0:0 inet dhcp

iface enp3s0 inet static
  address 192.168.3.100
  netmask 255.255.255.0
 ```

civerachb-cpr, 2021-03-12:
The best solution is to simply assign a static IP address to the VLP-16 on the 192.168.131.0/24 subnet.  The VLP-16 has a web interface you can use to configure its IP address; consult the sensor's manual for more information on that.

Whenever we ship customized Jackals with VLP-16 sensors on them we assign the lidar an address like 192.168.131.20 (or .21, or .22, depending on how many lidar sensors are on the robot).  This lets us simply connect the sensor to any available ethernet port on the robot, and launch the appropriate node(s).  If you set the static IP address like this then you do not need to make any changes to the `interfaces` file.

If giving the VLP-16 a static IP address on the 192.168.131.0/24 subnet isn't possible, the next-best option would be to add another subnet to the br0 bridge:

```
auto lo br0 br0:0 br0:1
iface lo inet loopback

# Bridge together physical ports on machine, assign standard Clearpath Robot IP.
iface br0 inet static
  bridge_ports regex eno1
  address 192.168.131.1
  netmask 255.255.255.0
  bridge_maxwait 0

iface br0:0 inet dhcp

iface br0:1 inet static
  address 192.168.3.1
  netmask 255.255.255.0
```

You may need to add `ifup br0:1` to `/etc/rc.local` to force the additional static interface to come up correctly on boot.  If `/etc/rc.local` does not exist, create it, put

```
#!/bin/bash
ifup br0:1
```

in it, and ensure it's executable by running `sudo chmod +x /etc/rc.local`

NithishkumarS, 2021-03-12:
That is very helpful. I will try that. But wouldn't it flood the channel with lidar scans, creating latency? How do you deal with that here?

civerachb-cpr, 2021-03-12:
You can specify a destination host for the VLP data, so that rather than broadcasting to the whole network the packets are only sent to the designated host, in this cast the PC running the ROS Master: `192.168.131.1`

NithishkumarS, 2021-03-15:
Thanks! I tried creating a second subnet in bridge0 and that didn't seem to work. But changing the vlp-16's ip to 192.168.131.0/24 worked like a charm. It would be great if you could also tell me where to place the launch of velodyne point cloud generation package, so that it start every time the robot bring-up happens.

civerachb-cpr, 2021-03-16:
All of the launch files that get run on launch are found in `/etc/ros/melodic/ros.d` (or `/etc/ros/kinetic/ros.d` if you're using Kinetic).

You can either copy the launch file directly into that folder, use a symbolic link, or use the `robot_upstart` package to install it:

```
rosrun robot_upstart install --job ros --augment velodyne_pointcloud/launch/VLP16_points.launch
```

After running that, the launch file should automatically appear in the `ros.d` folder and should launch when the robot boots, or whenever you run `sudo systemctl restart ros`

NithishkumarS, 2021-03-16:
That works great. Thanks!

closed 2021-03-16 by NithishkumarS