jackal/jackal_robot #29: /imu/data topic doesn't publish any data

NithishkumarS, 2021-03-19:
/imu/data_raw publishes data at 80Hz. But /imu/data has stopped publishing data. 
I checked the rosbags from previous runs and I could see the imu/data topic publishing data. I have restarted the robot a few times as well. /imu_filter node doesn't seem to publish it.

civerachb-cpr, 2021-03-22:
Is the `imu_filter` node actually running?  Even if it is, are there any warnings or errors in the ROS logs?  You can check these with `sudo journalctl -u ros`.

Have you edited the IMU parameters either by setting `JACKAL_CONTROL_EXTRAS_PATH` or by editing `robot_localization.yaml` or `mag_config_default.yaml`?

NithishkumarS, 2021-03-22:
Yeah, I can see the imu_filter in the rosnode list. 
I can also see other topics /imu/magnetic_field being published as a zero vector. 

When I ran journalctl -u ros command 
```
/imu_filter/gain: 0.1
/imu_filter/mag_bias_x: 0.0
/imu_filter/mag_bias_y: 0.0
/imu_filter/mag_bias_z: 0.0
/imu_filter/publish_tf: False
 /imu_filter/use_mag: True
/imu_filter/use_magnetic_field_msg: False
/imu_filter/zeta: 0.001
```
I could see the IMU_filter node setting its params. I didn't see any errors there. 

I have edited the robot_localization.yaml file. Here is my version of it. I don't see it affecting the imu_filter node. 
[dual_ekf_jackal.yaml](https://github.com/jackal/jackal_robot/files/6184334/dual_ekf_jackalyaml.txt)

After sshing into the robot, i tried ```echo $JACKAL_CONTROL_EXTRAS_PATH``` and it returned an empty string.

civerachb-cpr, 2021-03-22:
One thing I noticed in your dual_ekf file is that you've defined imu0 twice: once on line 46 and again on 153.  I'm assuming you intended one of those to be imu1?  In either case, I agree that that should not be affecting the imu_filter node.
Could you please run the following and attach the output?

```
rosnode info /imu_filter
rostopic info /imu/data_raw
rostopic info /imu/data
rosrun rqt_graph rqt_graph
```

For the last command, inside rqt_graph please ensure "Nodes/Topics (All)" is selected in the drop-down at the top.  You should be able to save the graph as an SVG to share it.

NithishkumarS, 2021-03-23:
rosnode info /imu_filter
```
Node [/imu_filter]
Publications: 
 * /imu/data [sensor_msgs/Imu]
 * /imu_filter/parameter_descriptions [dynamic_reconfigure/ConfigDescription]
 * /imu_filter/parameter_updates [dynamic_reconfigure/Config]
 * /rosout [rosgraph_msgs/Log]
 * /tf [tf/tfMessage]

Subscriptions: 
 * /imu/data_raw [sensor_msgs/Imu]

Services: 
 * /imu_filter/get_loggers
 * /imu_filter/set_logger_level
 * /imu_filter/set_parameters

contacting node http://hoali6100d:36340/ ...
Pid: 1497
Connections:
 * topic: /rosout
    * to: /rosout
    * direction: outbound
    * transport: TCPROS
 * topic: /tf
    * to: /ekf_localization
    * direction: outbound
    * transport: TCPROS
 * topic: /imu/data
    * to: /ekf_localization
    * direction: outbound
    * transport: TCPROS
 * topic: /imu/data_raw
    * to: /jackal_node (http://hoali6100d:34814/)
    * direction: inbound
    * transport: TCPROS
```

rostopic info /imu/data_raw
```
header: 
  seq: 78539
  stamp: 
    secs: 1616516679
    nsecs:  71629955
  frame_id: "imu_link"
orientation: 
  x: 0.0
  y: 0.0
  z: 0.0
  w: 0.0
orientation_covariance: [-1.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
angular_velocity: 
  x: -0.00103545425924
  y: 0.000108931868087
  z: 0.00245233235432
angular_velocity_covariance: [1.218467815533586e-07, 0.0, 0.0, 0.0, 1.218467815533586e-07, 0.0, 0.0, 0.0, 1.218467815533586e-07]
linear_acceleration: 
  x: 0.0132166640192
  y: -0.00580161829512
  z: 9.79571566347
linear_acceleration_covariance: [8.661248102725949e-06, 0.0, 0.0, 0.0, 8.661248102725949e-06, 0.0, 0.0, 0.0, 8.661248102725949e-06]
---
rostopic info /imu/data
```
  
```

NithishkumarS, 2021-03-23:
I reinstalled the OS, and it works fine now. But I would still like to find out what caused the problem initially so that I don't have to reinstall again.

civerachb-cpr, 2021-03-23:
It looks like you ran `rostopic echo` and not `rostopic info` for the `/imu/data` and `/imu/data_raw` topics.  I wanted to confirm what node(s) were subscribing to/publishing those topics, not see the actual data on the topic.  And you didn't include the requested rqt graph.  Without that information it's hard to make a determination as to what went wrong.

If reinstalling the OS worked my best guess is that a configuration file was either modified or got corrupted that wound up breaking a connection on one of the afore-mentioned topics.

NithishkumarS, 2021-03-24:
The reason I didn't post the rqt graph was that, I can't generate it (the new one works just fine and has a outgoing topic from imu_filter node) anymore since I reinstalled it. I understand that makes it almost impossible to debug this issue. Thanks.

closed 2021-03-24 by NithishkumarS