jackal/jackal_robot #31: Warnings on base.launch (Melodic)

RobertBlakeAnderson, 2021-06-10:
I receive some warnings when I manually bring up the robot in Melodic. Teleop via the PS controller still works.

```
administrator@cpr-aus12-1:~$ roslaunch jackal_base base.launch 
... logging to /home/administrator/.ros/log/73a7251e-ca26-11eb-b0b2-3c7c3f231ec2/roslaunch-cpr-aus12-1-2476.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

xacro: in-order processing became default in ROS Melodic. You can drop the option.
started roslaunch server http://cpr-aus12-1:40555/

SUMMARY
========

PARAMETERS
 * /bluetooth_teleop/joy_node/autorepeat_rate: 20
 * /bluetooth_teleop/joy_node/deadzone: 0.1
 * /bluetooth_teleop/joy_node/dev: /dev/input/ps4
 * /bluetooth_teleop/teleop_twist_joy/axis_angular: 0
 * /bluetooth_teleop/teleop_twist_joy/axis_linear: 1
 * /bluetooth_teleop/teleop_twist_joy/enable_button: 4
 * /bluetooth_teleop/teleop_twist_joy/enable_turbo_button: 5
 * /bluetooth_teleop/teleop_twist_joy/scale_angular: 1.4
 * /bluetooth_teleop/teleop_twist_joy/scale_linear: 0.4
 * /bluetooth_teleop/teleop_twist_joy/scale_linear_turbo: 2.0
 * /diagnostic_aggregator/analyzers/jackal/analyzers/general/expected: ['jackal_node: Ge...
 * /diagnostic_aggregator/analyzers/jackal/analyzers/general/path: General
 * /diagnostic_aggregator/analyzers/jackal/analyzers/general/remove_prefix: jackal_node
 * /diagnostic_aggregator/analyzers/jackal/analyzers/general/timeout: 5.0
 * /diagnostic_aggregator/analyzers/jackal/analyzers/general/type: diagnostic_aggreg...
 * /diagnostic_aggregator/analyzers/jackal/analyzers/sensors/expected: ['jackal_node: /i...
 * /diagnostic_aggregator/analyzers/jackal/analyzers/sensors/path: Sensor Topics
 * /diagnostic_aggregator/analyzers/jackal/analyzers/sensors/remove_prefix: jackal_node
 * /diagnostic_aggregator/analyzers/jackal/analyzers/sensors/timeout: 5.0
 * /diagnostic_aggregator/analyzers/jackal/analyzers/sensors/type: diagnostic_aggreg...
 * /diagnostic_aggregator/analyzers/jackal/analyzers/teleop/expected: ['bluetooth_teleo...
 * /diagnostic_aggregator/analyzers/jackal/analyzers/teleop/path: Bluetooth Joystick
 * /diagnostic_aggregator/analyzers/jackal/analyzers/teleop/remove_prefix: bluetooth_teleop/...
 * /diagnostic_aggregator/analyzers/jackal/analyzers/teleop/timeout: 5.0
 * /diagnostic_aggregator/analyzers/jackal/analyzers/teleop/type: diagnostic_aggreg...
 * /diagnostic_aggregator/analyzers/jackal/path: Jackal Base
 * /diagnostic_aggregator/analyzers/jackal/type: diagnostic_aggreg...
 * /ekf_localization/base_link_frame: base_link
 * /ekf_localization/frequency: 50
 * /ekf_localization/imu0: /imu/data
 * /ekf_localization/imu0_config: [False, False, Fa...
 * /ekf_localization/imu0_differential: False
 * /ekf_localization/odom0: /jackal_velocity_...
 * /ekf_localization/odom0_config: [False, False, Fa...
 * /ekf_localization/odom0_differential: False
 * /ekf_localization/odom_frame: odom
 * /ekf_localization/world_frame: odom
 * /imu_filter/gain: 0.1
 * /imu_filter/mag_bias_x: 0
 * /imu_filter/mag_bias_y: 0
 * /imu_filter/mag_bias_z: 0
 * /imu_filter/publish_tf: False
 * /imu_filter/use_mag: False
 * /imu_filter/zeta: 0.001
 * /jackal_joint_publisher/publish_rate: 50
 * /jackal_joint_publisher/type: joint_state_contr...
 * /jackal_node/port: /dev/jackal
 * /jackal_node/require/publishers: ['status', 'feedb...
 * /jackal_node/require/subscribers: ['cmd_drive', 'wi...
 * /jackal_node/wireless_interface: wlo1
 * /jackal_velocity_controller/angular/z/has_acceleration_limits: True
 * /jackal_velocity_controller/angular/z/has_velocity_limits: True
 * /jackal_velocity_controller/angular/z/max_acceleration: 25.0
 * /jackal_velocity_controller/angular/z/max_velocity: 4.0
 * /jackal_velocity_controller/cmd_vel_timeout: 0.25
 * /jackal_velocity_controller/enable_odom_tf: False
 * /jackal_velocity_controller/k_l: 0.1
 * /jackal_velocity_controller/k_r: 0.1
 * /jackal_velocity_controller/left_wheel: ['front_left_whee...
 * /jackal_velocity_controller/linear/x/has_acceleration_limits: True
 * /jackal_velocity_controller/linear/x/has_velocity_limits: True
 * /jackal_velocity_controller/linear/x/max_acceleration: 20.0
 * /jackal_velocity_controller/linear/x/max_velocity: 2.0
 * /jackal_velocity_controller/pose_covariance_diagonal: [0.001, 0.001, 10...
 * /jackal_velocity_controller/publish_rate: 50
 * /jackal_velocity_controller/right_wheel: ['front_right_whe...
 * /jackal_velocity_controller/twist_covariance_diagonal: [0.001, 0.001, 0....
 * /jackal_velocity_controller/type: diff_drive_contro...
 * /jackal_velocity_controller/wheel_radius_multiplier: 1.0
 * /jackal_velocity_controller/wheel_separation_multiplier: 1.5
 * /robot_description: <?xml version="1....
 * /rosdistro: melodic
 * /rosversion: 1.14.11
 * /twist_mux/locks: [{'topic': 'e_sto...
 * /twist_mux/topics: [{'topic': 'joy_t...

NODES
  /
    controller_spawner (controller_manager/spawner)
    diagnostic_aggregator (diagnostic_aggregator/aggregator_node)
    ekf_localization (robot_localization/ekf_localization_node)
    imu_filter (imu_filter_madgwick/imu_filter_node)
    jackal_node (jackal_base/jackal_node)
    robot_state_publisher (robot_state_publisher/robot_state_publisher)
    rosserial_message_info (rosserial_python/message_info_service.py)
    twist_marker_server (interactive_marker_twist_server/marker_server)
    twist_mux (twist_mux/twist_mux)
  /bluetooth_teleop/
    joy_node (joy/joy_node)
    teleop_twist_joy (teleop_twist_joy/teleop_node)
  /navsat/
    nmea_topic_driver (nmea_navsat_driver/nmea_topic_driver)

auto-starting new master
process[master]: started with pid [2507]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 73a7251e-ca26-11eb-b0b2-3c7c3f231ec2
process[rosout-1]: started with pid [2526]
started core service [/rosout]
process[robot_state_publisher-2]: started with pid [2548]
process[jackal_node-3]: started with pid [2549]
process[rosserial_message_info-4]: started with pid [2550]
process[navsat/nmea_topic_driver-5]: started with pid [2551]
process[imu_filter-6]: started with pid [2553]
process[controller_spawner-7]: started with pid [2564]
[ WARN] [1623355207.060551186]: Rejecting message on topicId=113, length=321 with bad checksum.
[ WARN] [1623355207.060700930]: Received message with unrecognized topicId (112).
[ WARN] [1623355207.063139214]: Received message with unrecognized topicId (113).
[ WARN] [1623355207.063283941]: Received message with unrecognized topicId (114).
[ WARN] [1623355207.063296558]: Received message with unrecognized topicId (112).
[ WARN] [1623355207.063445203]: Received message with unrecognized topicId (112).
process[ekf_localization-8]: started with pid [2567]
process[twist_mux-9]: started with pid [2576]
process[bluetooth_teleop/joy_node-10]: started with pid [2578]
process[bluetooth_teleop/teleop_twist_joy-11]: started with pid [2579]
process[twist_marker_server-12]: started with pid [2592]
process[diagnostic_aggregator-13]: started with pid [2596]
[ WARN] [1623355207.367760966]: Rejecting message on topicId=113, length=321 with bad checksum.
```

civerachb-cpr, 2021-06-10:
Those warnings are relatively common and benign when the rosserial node starts up.  Sometimes there's excess noise on the serial line during startup or high bandwidth during initialization can cause some messages to be truncated.  As long as these messages are only showing up briefly when the nodes are starting and aren't persistently appearing in the logs it's fine.

If you run `sudo journalctl -fu ros` after rebooting the robot you may see similar warnings in the systemd job's logs.

When manually launching the nodes are you stopping the ROS systemd job by running `sudo systemctl stop ros` before running `roslaunch jackal_base base.launch`?

RobertBlakeAnderson, 2021-06-11:
Yes, I'm running ```sudo systemctl stop ros.service``` before doing this. I basically want to reconfig my robot to not do the automatic upstart. I take it that there are some side effects in the systemd job that are necessary to config the robot, so by not running it I produce these warnings. Here's another batch of output from trying it again today.

```
auto-starting new master
process[master]: started with pid [2636]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to fad2fd0e-cad2-11eb-80df-3c7c3f231ec2
process[rosout-1]: started with pid [2655]
started core service [/rosout]
process[robot_state_publisher-2]: started with pid [2677]
process[jackal_node-3]: started with pid [2678]
process[rosserial_message_info-4]: started with pid [2679]
process[navsat/nmea_topic_driver-5]: started with pid [2680]
process[imu_filter-6]: started with pid [2684]
process[controller_spawner-7]: started with pid [2694]
[ WARN] [1623429307.270669760]: Rejecting message on topicId=112, length=69 with bad checksum.
process[ekf_localization-8]: started with pid [2698]
process[twist_mux-9]: started with pid [2705]
[ WARN] [1623429307.282658897]: Received message with unrecognized topicId (113).
[ WARN] [1623429307.282682884]: Received message with unrecognized topicId (114).
[ WARN] [1623429307.282797325]: Received message with unrecognized topicId (112).
[ WARN] [1623429307.282846010]: Received message with unrecognized topicId (112).
process[bluetooth_teleop/joy_node-10]: started with pid [2707]
process[bluetooth_teleop/teleop_twist_joy-11]: started with pid [2714]
process[twist_marker_server-12]: started with pid [2721]
[ WARN] [1623429307.300516466]: Received message with unrecognized topicId (113).
[ WARN] [1623429307.300552698]: Received message with unrecognized topicId (114).
[ WARN] [1623429307.300632048]: Received message with unrecognized topicId (111).
[ WARN] [1623429307.300698766]: Received message with unrecognized topicId (110).
process[diagnostic_aggregator-13]: started with pid [2730]
[ WARN] [1623429307.306660458]: Received message with unrecognized topicId (113).
[ WARN] [1623429307.306695970]: Received message with unrecognized topicId (114).
[ WARN] [1623429307.318647691]: Received message with unrecognized topicId (113).
[ WARN] [1623429307.318675747]: Received message with unrecognized topicId (114).
[ WARN] [1623429307.330652106]: Received message with unrecognized topicId (113).
[ WARN] [1623429307.330674578]: Received message with unrecognized topicId (114).
[ WARN] [1623429307.342655067]: Received message with unrecognized topicId (113).
[ WARN] [1623429307.342673619]: Received message with unrecognized topicId (114).
[ WARN] [1623429307.354229459]: Received message with unrecognized topicId (110).
[ WARN] [1623429307.360683732]: Received message with unrecognized topicId (113).
[ WARN] [1623429307.360707752]: Received message with unrecognized topicId (114).
[ WARN] [1623429307.372693562]: Received message with unrecognized topicId (113).
[ WARN] [1623429307.372718887]: Received message with unrecognized topicId (114).
[ WARN] [1623429307.372807082]: Received message with unrecognized topicId (112).
[ WARN] [1623429307.372994893]: Received message with unrecognized topicId (112).
[ WARN] [1623429307.373010144]: Received message with unrecognized topicId (112).
[ WARN] [1623429307.373030817]: Received message with unrecognized topicId (112).
[ WARN] [1623429307.384653334]: Received message with unrecognized topicId (113).
[ WARN] [1623429307.384675960]: Received message with unrecognized topicId (114).
[ WARN] [1623429307.396341294]: Received message with unrecognized topicId (110).
[ WARN] [1623429307.396776951]: Received message with unrecognized topicId (113).
[ WARN] [1623429307.396786887]: Received message with unrecognized topicId (114).
[ WARN] [1623429307.414521282]: Received message with unrecognized topicId (113).
[ WARN] [1623429307.414542083]: Received message with unrecognized topicId (114).
[ WARN] [1623429307.420666006]: Received message with unrecognized topicId (113).
[ WARN] [1623429307.420680527]: Received message with unrecognized topicId (114).
[ WARN] [1623429307.438728272]: Received message with unrecognized topicId (113).
[ WARN] [1623429307.438744388]: Received message with unrecognized topicId (114).
[ WARN] [1623429307.444393324]: Received message with unrecognized topicId (110).
[ WARN] [1623429307.450732353]: Received message with unrecognized topicId (113).
[ WARN] [1623429307.450747561]: Received message with unrecognized topicId (114).
[ WARN] [1623429307.462769790]: Received message with unrecognized topicId (113).
[ WARN] [1623429307.462832175]: Received message with unrecognized topicId (114).
[ WARN] [1623429307.462924552]: Received message with unrecognized topicId (112).
[ WARN] [1623429307.462974467]: Received message with unrecognized topicId (112).
[ WARN] [1623429307.480842097]: Received message with unrecognized topicId (113).
[ WARN] [1623429307.480908126]: Received message with unrecognized topicId (114).
[ WARN] [1623429307.480944759]: Received message with unrecognized topicId (112).
[ WARN] [1623429307.480992176]: Received message with unrecognized topicId (112).
[... 178 lines cut ...]
[ WARN] [1623429309.323740929]: Received message with unrecognized topicId (111).
[ WARN] [1623429309.323756855]: Received message with unrecognized topicId (110).
[ WARN] [1623429309.352421939]: Received message with unrecognized topicId (110).
[... cut ...]
[ WARN] [1623429310.156516883]: Received message with unrecognized topicId (110).
[ WARN] [1623429310.210508035]: Received message with unrecognized topicId (110).
[ WARN] [1623429310.258364089]: Received message with unrecognized topicId (110).
[ WARN] [1623429310.271597358]: Missing connection to topic status, required by parameter ~require/publishers
[ WARN] [1623429310.271609186]: Connected client failed to establish the publishers and subscribers dictated by require parameter. Re-requesting topics.

```

civerachb-cpr, 2021-06-28:
That's weird.  I've tried replicating this on one of the Jackals we have in the office and wasn't able to.
Are you seeing these same errors when you let the systemd job start?  (i.e. do they appear in `sudo journalctl -fu ros`, or do they only occur when the systemd job is inactive and you launch the jackal_base manually?)

closed 2022-11-08 by tonybaltovski