jackal/jackal_robot #35: apt-get install ros-melodic-jackal-robot broken

ObjExtract, 2021-07-17:
All of a sudden apt-get install ros-melodic-jackal-robot fails. Getting "no package".

civerachb-cpr, 2021-07-19:
A few questions to help us resolve this for you:

What version of Ubuntu are you using?

Are you using Clearpath's Ubuntu ISO to install the OS on the robot, or are you using a different Ubuntu installer (e.g. Ubuntu Desktop, Nvidia's SDK Manager [for a Jetson])?

If you're not using one of Clearpath's ISOs, please make sure that you're adding the official ROS apt sources, as well as Clearpath's:
- http://wiki.ros.org/melodic/Installation/Ubuntu
- https://packages.clearpathrobotics.com

civerachb-cpr, 2021-09-29:
It's been over two months with no reply.  I'm hoping you've gotten this issue resolved.  If it's still causing problems please leave a comment and we can re-open the issue.

closed 2021-09-29 by civerachb-cpr