jackal/jackal_robot #36: FrameID for IMU data

RobertBlakeAnderson, 2021-08-19:
I'm trying to figure out how to change the frame of IMU data coming from the MCU. As I understand it, the MCU launches a process on startup which publishes the IMU topic, among others. Then `jackal_node` sets up a serial connection to the MCU, then republishes that data on the onboard PC.

I want to basically change the frame ID of the IMU data from `imu_link` to `<namespace>/imu_link`. This is to support multirobot applications where every robot has an `imu_link`. Is there a simple way to do this? I'm guessing I may need to modify the upstart job on the MCU somehow.

RobertBlakeAnderson, 2021-08-19:
Possible solution - add a `imu_frame` ROS param to the `imu_filter_madgwick` node. With that param specified, the filter node can overwrite the frame of the incoming raw messages prior to republishing them. I'm willing to implement this if it sounds good.

civerachb-cpr, 2021-08-19:
My concern with that is that the `imu/data_raw` topic would still have data being published with the global `imu_link` frame ID; not sure if that might cause problems or not with the TF; I'd need to test it to be sure.

How are you setting the namespace for each robot?  Just setting the `ROS_NAMESPACE` envar?

RobertBlakeAnderson, 2021-08-19:
We simply set the namespace in our launch file, either using <group> tags or the "ns" property of the <node> tag. The result is that the MCU topics are namespaced the same as `jackal_node`, i.e. `/jackal/imu/data_raw' is published by `/jackal/jackal_node`.

The problem we're trying to solve is that, right now, the IMU messages with frame `imu_link` are being sent to our `robot_localization` EKF node. EKF is then unable to transform from `imu_link` to `jackal/base_link` because the namespaces are different.

What I propose is to have madgwick add any necessary namespacing to the frame_id field. This could be a ROS param string that overwrites the field, or perhaps we just prepend the namespace of the madgwick node to the raw frame_id.

RobertBlakeAnderson, 2021-08-19:
And we're not using the embedded GPS right now, but it occurs to me that its messages may have the same problem. So rather than using madgwick specifically, one might create a new node class that listens to both the GPS and IMU messages and republishes each with properly namespaced frames.

RobertBlakeAnderson, 2021-09-21:
I've implemented a fix for this in #38

civerachb-cpr, 2021-09-29:
Rather than using a single global master with every robot running inside its own namespace, have you looked into a multi-master system: http://wiki.ros.org/multi_jackal_tutorials.