jackal/jackal_robot #39: Reading the Encoders in JackalHardware

RobertBlakeAnderson, 2021-09-21:
We are experiencing an issue related to the odometry produced by JackalHardware. The encoder positions are simply copied into the joint position values (see JackalHardware::copyJointsFromHardware()). This means that the robot travel is preserved when restarting the base node. Even rebooting the top computer doesn't reset the odometry, since the encoder states are stored on the MCU. The result is that, on repeat launches, the EKF loads old odometry information, so the robot jumps to incorrect coordinates.

Contrast this with how things are done on the Husky: https://github.com/husky/husky/blob/noetic-devel/husky_base/src/husky_hardware.cpp#L146
With Husky, the joint positions are initialized to zero, and then _changes_ in the encoder readings are accumulated into the positions. Thus, we don't have the above problem on our Husky.

I would recommend adopting the Husky methodology for Jackal. I'm willing to implement and PR this you are in agreement.

civerachb-cpr, 2021-09-29:
To make sure I understand the problem you're describing:
1) Start up the robot, odom is (0, 0)
2) Drive the robot around to (X, Y)
3) Restart the `jackal_base` node. The odom frame remains (X, Y), instead of resetting to (0, 0)

Is this actively causing problems when operating the robot, or is it merely an annoyance/theoretical problem that the robot may not always start directly above the odom frame's origin?

RobertBlakeAnderson, 2021-09-29:
Yes, you have it right. It's an annoyance, but a pretty big one we're finding out. The problem is the robot is not just starting in arbitrary odom coordinates, but also arbitrary _map_ coordinates. EKF is seeing that jump in the odometry and interpreting it as motion. When doing repeated testing on the robot, we had to fully shut down the platform in order to reset the state information on the motor controller. With the fork I've made, that's no longer necessary.

Tobias-Fischer, 2022-03-02:
Hi @RobertBlakeAnderson - any chance you could open a PR with this change?

RobertBlakeAnderson, 2022-03-03:
Happy to, if Clearpath is willing to merge it. @civerachb-cpr ?

In the meantime, [this](https://github.com/RobertBlakeAnderson/jackal_robot/commit/f85ec38b5ace5ded52d91b4557f5ee0d1c954df3) commit from our branch shows how I fixed it.

civerachb-cpr, 2022-03-03:
@RobertBlakeAnderson we're always happy to review PRs.  A quick look at the commit you linked to looks okay, but before merging it I'd probably want to get someone from our autonomy team to test it out to make sure it doesn't introduce any problems with our indoor & outdoor autonomy packages.  (I personally doubt it would, but I'd want to get a second, more-qualified set of eyes on it to be sure.)

tonybaltovski, 2022-03-03:
@RobertBlakeAnderson your commit looks good to me.  I would prefer a C++03 style given the rest of the code is that.

RobertBlakeAnderson, 2022-03-03:
I take it you mean `std::isnan`? That can certainly be changed to the old macro version.

tonybaltovski, 2022-03-04:
I mean using `std::numeric_limits<double>::quiet_NaN()` is declared in `<limits>` instead of `NAN`.  Also, adding `<cmath>` for  `std::isnan()`.   Could this be separate PR?  The IMU frame changes are very specific to your use-case.  Thanks!

RobertBlakeAnderson, 2022-03-15:
Thanks for the feedback @tonybaltovski . I've made that change and created a PR. I now have one for the odometry fix and one for the frame_id namespacing. I reworked my fork so that the two PRs have been separated into different source branches.

closed 2022-05-24 by RobertBlakeAnderson