jackal/jackal_robot #46: PS4 joystick does not work, after kernel update

adamanov, 2022-02-22:
**Please provide the following information:**
 - OS: (e.g. Ubuntu 18.04)
 - ROS Distro: (e.g. Melodic)
 - Build with ISO image. 
 - `sudo apt-get install ubuntu-desktop` # To get GUI version of jackal PC ubuntu 

 **Actual behaviour**

We had a problem with our Ethernet and WLAN network, therefore we had to update our kernel from 4.15 to 5.. 
As soon as we update, our WLAN starts working as expected, however, we lose control with Joystick, it does not work anymore.
We paired it again, the joystick commands are published on certain topics, and we can see them through `rostopic echo ...` 
However, the `jackal_node` does not make the robot move. 

 A clear and concise description of what you encountered.

**To Reproduce**
Provide the steps to reproduce:
1. install ISO image with Melodic on jackal, 
2. `sudo apt-get install ubuntu-desktop` 
3. `uname -sr` # 4... 
3. ` sudo apt-get install --install-recommends linux-generic-hwe-18.04 xserver-xorg-hwe-18.04  `
4. ` sudo reboot` 
5. `uname –sr` # 5..
6. Jackal does not move anymore

civerachb-cpr, 2022-02-22:
Sorry to hear you're having problems with your Jackal!
Can you please run a couple of quick tests to help diagnose the issue?

1) Verify that the low-level messages are being received by the kernel:
```bash
sudo apt-get install jstest-gtk
jstest /dev/input/ps4
```
You should see a stream of output indicating the states of the buttons and axes.  If you do not, then this would indicate that there could be a problem with the `ds4drv` driver we use for the PS4 controllers.

2) Verify that the `bluetooth_teleop/joy` node is publishing correctly:
```bash
sudo systemctl restart ros
rostopic echo /bluetooth_teleop/joy
```
Again, make sure that button presses and joystick motions are reflected properly in the output.  Also take note of the indices of the vertical & horizontal axes of the left stick and the indices of the left & right shoulder buttons.  The ROS driver specifies the indices for the deadman switches & control axes, so if these have changed as a result of the kernel update you may need to remap the controls.

3) Make sure the motor controller itself is working:
```bash
rostopic pub /cmd_vel geometry_msgs/Twist "linear:
  x: 0.1
  y: 0.0
  z: 0.0
angular:
  x: 0.0
  y: 0.0
  z: 0.0" -r 10
```
This will drive the robot forwards at 10cm/s, so make sure the robot is on blocks or has a clear path forwards. Cancelling will stop the robot.
If the robot doesn't move at all then there may be a problem with the communication between the PC and the MCU, or between the MCU and the motors.  Either is unrelated to the bluetooth controller.

4) Verify that `bluetooth_teleop/cmd_vel` is publishing correctly:
```bash
rostopic echo /bluetooth_teleop/cmd_vel
```
Hold down the left shoulder button and move the left thumb stick.  You should see the Linear X and Angular Z parameters changing.  As with the previous test, the robot should move during this procedure, so make sure it's not going to run into anything.

Some additional output that will be helpful for diagnosing possible MCU issues:
```bash
rostopic echo /status
ls -l /dev
groups
sudo udevadm info --attribute-walk /dev/jackal
```

adamanov, 2022-02-23:
> Sorry to hear you're having problems with your Jackal! Can you please run a couple of quick tests to help diagnose the issue?
> 
>     1. Verify that the low-level messages are being received by the kernel:
> 
> 
> ```shell
> sudo apt-get install jstest-gtk
> jstest /dev/input/ps4
> ```
> 
> You should see a stream of output indicating the states of the buttons and axes. If you do not, then this would indicate that there could be a problem with the `ds4drv` driver we use for the PS4 controllers.
> 
>     2. Verify that the `bluetooth_teleop/joy` node is publishing correctly:
> 
> 
> ```shell
> sudo systemctl restart ros
> rostopic echo /bluetooth_teleop/joy
> ```
> 
> Again, make sure that button presses and joystick motions are reflected properly in the output. Also take note of the indices of the vertical & horizontal axes of the left stick and the indices of the left & right shoulder buttons. The ROS driver specifies the indices for the deadman switches & control axes, so if these have changed as a result of the kernel update you may need to remap the controls.
> 
>     3. Make sure the motor controller itself is working:
> 
> 
> ```shell
> rostopic pub /cmd_vel geometry_msgs/Twist "linear:
>   x: 0.1
>   y: 0.0
>   z: 0.0
> angular:
>   x: 0.0
>   y: 0.0
>   z: 0.0" -r 10
> ```
> 
> This will drive the robot forwards at 10cm/s, so make sure the robot is on blocks or has a clear path forwards. Cancelling will stop the robot. If the robot doesn't move at all then there may be a problem with the communication between the PC and the MCU, or between the MCU and the motors. Either is unrelated to the bluetooth controller.
> 
>     4. Verify that `bluetooth_teleop/cmd_vel` is publishing correctly:
> 
> 
> ```shell
> rostopic echo /bluetooth_teleop/cmd_vel
> ```
> 
> Hold down the left shoulder button and move the left thumb stick. You should see the Linear X and Angular Z parameters changing. As with the previous test, the robot should move during this procedure, so make sure it's not going to run into anything.
> 
> Some additional output that will be helpful for diagnosing possible MCU issues:
> 
> ```shell
> rostopic echo /status
> ls -l /dev
> groups
> sudo udevadm info --attribute-walk /dev/jackal
> ```

Hello, thank you for your quick response.

I went today to Lab to test your proposed solutions. But after turning on the jackal, he started moving like nothing happened :) 
We assume that it had some issue with MCU, but hope it will not pup up again 

Thanks again, all best

closed 2022-02-23 by adamanov