jackal/jackal_robot #60: Running calibrate_compass node from jackal_base package results in attribute error

sehv01, 2022-11-08:
- OS: Ubuntu 20.04
 - ROS Distro: noetic
 - Built from source or installed: installed
 - Package: ros-noetic-jackal-robot
 - Version: 0.7.4-1focal.20220526.195017

 **Expected behaviour**
Running the jackal_base package results in jackal being calibrated.

 **Actual behaviour**
Script being unable to compute calibration.

**To Reproduce**
Provide the steps to reproduce:
1. ssh into jackal
2. execute rosrun jackal_base calibrate_compass
3. see the output saying: "Unable to compute calibration from recorded bag"

**Other notes**
Terminal output running calibrate_compass node:
```administrator@cpr-j100:~$ rosrun jackal_base calibrate_compass
Started rosbag record, duration 60 seconds, pid [1408]
Started motion commands, pid [1409]
[ INFO] [1665751505.761305011]: Subscribing to /imu/data_raw
[ INFO] [1665751505.765065728]: Subscribing to /imu/mag
[ INFO] [1665751505.767817117]: Subscribing to /imu/rpy/raw
[ INFO] [1665751505.770318908]: Subscribing to /tf
[ INFO] [1665751505.773660499]: Recording to '/tmp/calibrate_compass.e91S/imu_record.bag'.
Test underway.
Time remaining: 0
Shutting down motion command publisher.
Waiting for rosbag to shut down.
Computing magnetic calibration.
Unable to compute calibration from recorded bag.
Output in /tmp/compute_output.log
```

Output from compute_output.log:
```administrator@cpr-j100:~$ cat /tmp/compute_output.log
Traceback (most recent call last):
  File "/opt/ros/noetic/lib/jackal_base/compute_calibration", line 42, in <module>
    vecs.append((float(msg.vector.x), float(msg.vector.y), float(msg.vector.z)))
AttributeError: '_sensor_msgs__MagneticField' object has no attribute 'vector'
```

sehv01, 2022-11-08:
pull request #59

tonybaltovski, 2022-11-08:
Thanks for the PR!

closed 2022-11-08 by tonybaltovski