jackal/jackal_robot #65: Compass calibration for Foxy

mllofriu, 2023-02-09:
The ROS1 version of this software had a utility to calibrate the Compass - https://www.clearpathrobotics.com/assets/guides/melodic/jackal/calibration.html.

Is there an existing alternative for Foxy and forward versions?

I was actually looking for something to calibrate the accelerometer, as it seems to be throwing unexpected values with Foxy (e.g. no clear gravity vector). Any guidance on that front will be welcomed too.

tonybaltovski, 2023-02-09:
Hi @mllofriu, we are looking into this.  It seems like a bug in the firmware that is removing gravity.

mllofriu, 2023-02-09:
Thanks for the quick response. Do you want me to open a different one regarding compass calibration?

tonybaltovski, 2023-02-09:
Sure, that would be great! Thanks!

mllofriu, 2023-02-15:
Just for the sake of our own planning, do you have an estimated time to resolve this? Can we help? I think the firmware is not public, but if you want to share, we can take a look.

tonybaltovski, 2023-03-24:
Thanks for your patience on this.  Version 1.0.2 of the firmware has the fix for this.  Let me know if you run into any issues.