jackal/jackal_robot #67: One to three seconds delay in /imu/data_raw messages

mllofriu, 2023-03-31:
**Please provide the following information:**
 - OS: Tegra 20.04 5.10.65-tegra
 - ROS Distro: Foxy
 - Built from source or installed: installed ros, packages from source as in tutorial - https://clearpathrobotics.com/assets/guides/foxy/jackal/JackalInstallRobotSoftware.html
 - Package version: ee4e3f68a9e0a7740eec058f85432bebbc7eba23

 **Expected behaviour**
The delay in /imu/data_raw messages is around the publishing period.

 **Actual behaviour**
The delay is above one second, and between 1-3 seconds most of the time.

**To Reproduce**
Provide the steps to reproduce:
1. Install firmware 1.0.2
2. ros2 launch jackal_robot bringup.launch.py
3. ros2 topic delay /imu/data_raw

The output is something like 

ros2 topic delay /imu/data_raw -w 50
average delay: -1.810
	min: -3.333s max: 0.972s std dev: 1.76784s window: 49
average delay: -0.350
	min: -1.039s max: -0.014s std dev: 0.46913s window: 50
average delay: -1.349
	min: -2.039s max: -1.012s std dev: 0.46911s window: 50
average delay: -2.349
	min: -3.039s max: -2.011s std dev: 0.46992s window: 50
average delay: -2.833
	min: -4.038s max: 0.266s std dev: 1.20963s window: 50
average delay: -0.057
	min: -0.743s max: 0.274s std dev: 0.46662s window: 50

The delay of /diagnostics (stamped on the same system), for comparison:
ros2 topic delay /diagnostics -w 50
average delay: 0.003
	min: 0.001s max: 0.005s std dev: 0.00170s window: 2
average delay: 0.003
	min: 0.001s max: 0.005s std dev: 0.00148s window: 5
average delay: 0.003
	min: 0.001s max: 0.005s std dev: 0.00142s window: 8
average delay: 0.002
	min: 0.001s max: 0.005s std dev: 0.00136s window: 11
average delay: 0.003
	min: 0.001s max: 0.005s std dev: 0.00134s window: 14
average delay: 0.003
	min: 0.001s max: 0.005s std dev: 0.00133s window: 17

**Other notes**
I noticed this while trying to run LIO-SAM over a bag recorded onboard. The algorithm keeps skipping LIDAR frames because it doesn't find IMU data around the same timestamp. I know the algorithms should be responsible of synchronizing the data, but still 1-3 seconds seemed a bit too much.

Could the clock in the firmware be out of sync with the one in the SBC? Is the /imu/data_raw message stamped in the microros component?

tonybaltovski, 2023-04-04:
Is the delay on just the IMU topic or all of the topics coming from the MCU?

mllofriu, 2023-04-04:
I didn't check other MCU topics, I will as soon as I can.

tonybaltovski, 2023-04-19:
@mllofriu we were able to reproduce the issue and the fix is in `1.0.3`.  Please let me know if you have any other problems.

mllofriu, 2023-04-21:
I installed the latest firmware and the problem seems to be fixed

> ros2 topic delay /imu/data_raw -w 50
average delay: 0.020
	min: 0.011s max: 0.032s std dev: 0.00589s window: 49
average delay: 0.021
	min: 0.012s max: 0.037s std dev: 0.00622s window: 50
average delay: 0.020
	min: 0.010s max: 0.040s std dev: 0.00671s window: 50
average delay: 0.020
	min: 0.012s max: 0.038s std dev: 0.00671s window: 50
average delay: 0.026
	min: 0.010s max: 0.045s std dev: 0.00944s window: 50
average delay: 0.022
	min: 0.010s max: 0.042s std dev: 0.00780s window: 50
average delay: 0.021
	min: 0.011s max: 0.037s std dev: 0.00652s window: 50
average delay: 0.020
	min: 0.011s max: 0.037s std dev: 0.00719s window: 50

Thanks again!

closed 2023-04-27 by tonybaltovski