jackal/jackal_robot #70: Magnetometer data is missing the covariance

mllofriu, 2023-05-02:
**Please provide the following information:**
 - OS: Ubuntu 20.04 in Tegra
 - ROS Distro: Foxy
 - Built from source or installed: Installed, built jackal packages from source
 - Package version: ee4e3f6

 **Expected behaviour**
The messages in /imu/mag have non-zero covariaces.

 **Actual behaviour**
The data has the Zero matrix as covariance.

**To Reproduce**
Provide the steps to reproduce:
1. run jackal bringup
2. ros2 topic echo /imu/mag

tonybaltovski, 2023-05-04:
@mllofriu The magnetometer didn't have covariance data from ROS 1 when it was only a Vector3 message.  Do you have a specific use case for the data?  We can look for the covariance for the sensor and add it to the next release of the firmware.

mllofriu, 2023-05-09:
Some lidar SLAM systems require a 9-degree imu, which could be composed of the 6-degree imu and the magnetometer. 

To be honest, I'm not sure how sensitive to the covariance any of these solutions (if at all) are:
 - use a (stateless?) madgwick filter to get a 9-degree estimate from both sensors, and use that as the 9-degree imu
 - for the algorithms allowing to read from /imu/mag directly, how sensitive they are to the reported covariance

All in all, I think this one is lower priority, unless you believe it affects the result of the /imu/data estimate (madgwick).