jackal/jackal_robot #72: The z axis of the imu (raw) flips at certain times

mllofriu, 2023-05-02:
**Please provide the following information:**
 - OS: Ubuntu 20.04 in Tegra
 - ROS Distro: Foxy
 - Built from source or installed: Installed, built jackal packages from source
 - Package version: ee4e3f6

 **Expected behaviour**
The messages in /imu/data_raw have a z component that is always pointing down while navigating in a plane.

 **Actual behaviour**
Large positive values are observed in the z component.
[image]

**To Reproduce**
Provide the steps to reproduce:
1. run jackal bringup
2. run the robot in a 3m radius circle
3. ros2 topic echo /imu/data_raw --csv
4. plot with R

```
require('reshape2')

read_csv_and_groom <-function(path, cols_to_select, source) 
{
  data <- read.csv(path, header = FALSE)
  data <- data[,cols_to_select]
  colnames(data) <- c('ts', 'ns','x', 'y', 'z')
  data$ns <- as.integer(data$ns)
  data$ts <- as.integer(data$ts)
  # transform secs rel to start
  data$ts <- data$ts - data$ts[1]
  # keep only milliseconds
  data$ms <- data$ns / 1e6
  print(data$ns)
  print(data$ms)
  data$ms <- data$ts * 1e3 + data$ms

  data <- melt(data, na.rm = TRUE, id.vars = 'ms', measure.vars = c('x', 'y', 'z'))
  data$source <- source
  
  data
}

imu <- read_csv_and_groom('~/Downloads/imu.csv', c(1,2,17,18,19), 'imu')
mag <- read_csv_and_groom('~/Downloads/mag.csv', c(1,2,4,5,6), 'mag')

all <- rbind(imu, mag)

require('ggplot2')
ggplot(all[all$source == 'imu',], aes(x = ms)) +
  geom_point(aes(y = value, color=interaction(source, variable))) 

```

tonybaltovski, 2023-05-04:
The accelerometer when stationary is measuring the force of gravity.  If the platform is not perfectly aligned with Earth's gravity vector it could have X or Y direction offsets.  Is the Jackal started on a flat surface?  Does the filtered data have the same offset?

marco-ambrosio, 2023-05-25:
I would like to add that the data published on  /Imu/data_raw are without covariance, thus standard  filter implemented in ROS can do pretty nothing on them